lina-usc/pylossless #8: pad flags?

scott-huberty, 2022-10-03:
Not sure what the purpose of  `padflags` is.. @jadesjardins @andesha care to rack your brains?  : ) 

It is used in both [s01](https://github.com/BUCANL/EEG-IP-L/blob/2d42fe75ab9d07437b77a5e3e86540063e2ed3a3/code/scripts/s01_scalpart.htb#L149) and [s03](https://github.com/BUCANL/EEG-IP-L/blob/2d42fe75ab9d07437b77a5e3e86540063e2ed3a3/code/scripts/s01_scalpart.htb#L149).

It is from James [Vised Marks repo](https://github.com/BUCANL/Vised-Marks/blob/3a788555bdc7e2e44810e6c5ae893e70aa01d77c/marks/misc_func/padflags.m#L4)

MATLAB code..

```
% Function which given an array 'flags' of values, prepends and appends a value
% around a given nonzero block of data in the given array.
% This value can be customized via the vararg 'value'. (e.g. 'value',0.5)
function flags=padflags(EEG,flags,npad,varargin)

g=struct(varargin{:});

try g.value; catch, g.value=1; end;

for np=1:npad;
    for i=1:size(flags,3)-1;
        %forward...
        if any(flags(:,:,i+1)) && ~any(flags(:,:,i))
            flags(:,:,i)=g.value;
        end
        %backward...
        if any(flags(:,:,(EEG.trials-(i-1))-1)) && ~any(flags(:,:,EEG.trials-(i-1)))
            flags(:,:,EEG.trials-(i-1))=g.value;
        end
    end
end
```

scott-huberty, 2022-10-03:
@jadesjardins Ok I see - so this  "value" is the colour/opacity of the annotation when viewing it during qcr?

christian-oreilly, 2022-10-03:
I don't see the relationship between color/opacity and annotation time duration (and the padding of it)...

Andesha, 2022-10-03:
[image]

In this image, the green arrow is pointing to an annotation that makes some period of time as rejected. In order to extend these a bit so that they aren't as abrupt, we created the "pad flags" portion which is pointed at by the red arrow. You can see that the opacity is different such that the padding (pad flags) around the annotation are more transparent.

Hope this helps!

Andesha, 2022-10-03:
Not really - it doesn't particularly change anything. I don't think anyone has ever strictly made a decision during QC on the padding...

Honestly, it's probably fine to package it as one annotation as you've outlined above :)

scott-huberty, 2022-10-03:
As a user I liked being able to see the original annotation and its paddding, but as Tyler said I never actually made a decision based on the pad. 

I **don't like** the idea of having two overlapping sets of annotations (one set being the flagged time, the other set being the flagged time + padded time) _just_ to be able to replicate this functionality. I could live without it : )

jadesjardins, 2022-10-03:
If MNE-Python is storing the time annotations like "events" (label, start, duration) .. which I think is great... I would store the pads with a different label. We don't need to lose the distinction between the flag and pad... EEGLAB was storing this a data channel so giving it the same name and a different value reduced the number of channels needed to describe the data. storing the pad annotations with a different label does not increase the size of the data structure is they are stored as "events".

scott-huberty, 2023-03-07:
Closing this issue since we pylsosless now pads small gaps bt bad annotations.

closed 2023-03-07 by scott-huberty