lina-usc/pylossless #9: Port Staging Script

scott-huberty, 2022-10-03:
The [Staging script](https://github.com/BUCANL/EEG-IP-L/blob/master/code/scripts/face13_staging.m) of the MATLAB pipeline is a study-specific script where line noise filtering is done and where time during experiment breaks are flagged to be ignored during the pipeline. 

the code to find task breaks is the important part that needs to be converted to python - maybe a function like `find_breaks`. The line noise filtering can just be done in the filtering step of S01.

scott-huberty, 2023-03-07:
Well - we still need to build/document functionality to use a staging script - but this issue specifically asks for a feature to find breaks - which we have. Shall I close? I can open another issue for a staging script feature (probably not for 0.01 though).

christian-oreilly, 2023-03-07:
Yes. The functionalities have been ported, albeit not using a staging script. Using a staging script might not be a very good idea anyway (due to code vulnerability to hacking). Let's consider this questions further if and when a use case present itself. For now, I agree with closing.

scott-huberty, 2023-03-07:
> ). Let's consider this questions further if and when a use case present itself.

+1

closed 2023-03-07 by scott-huberty