lina-usc/pylossless #31: Feature Request: Improve annotation drawing on timeseries plots.

scott-huberty, 2023-02-20:
The current method for manually creating ``annotations`` on time-series graphs, via click and drag, is clunky and non-intuitive.

It appears that there may well be a nicer solution. Check out this notebook I created.

https://colab.research.google.com/drive/18x3pawCqj6IU-cQQ5uDiKReLWYYaQrc4?usp=sharing

The solution was inspired from[ this stack overflow answer](https://stackoverflow.com/questions/71196269/dash-plotly-draw-vrect-in-graph-or-change-drawn-rectangles-height) and [this section of the Dash documentation](https://plotly.com/python/shapes/#drawing-shapes-with-a-mouse-on-cartesian-plots)

scott-huberty, 2023-02-20:
UPDATE: I created a Minimally Working example in [my Colab notebook](https://colab.research.google.com/drive/18x3pawCqj6IU-cQQ5uDiKReLWYYaQrc4?usp=sharing) of how this would work in a `dash` app.

FYI Make sure you allow 3rd party cookies in your browser if you want `JupyterDash` to work in Colab. It took me a while to figure that out....

IMO, a big benefit to using this build-in `Plotly` functionality to draw shapes, is that deleting shapes is already packaged in the functionality. This (deleting shapes) is something we haven't implemented in our current dash app, and it is something that we definitely need to support..

As we wrap up our current dash app, I am going to continue to probe on a separate dev branch.. I think it's definitely worth exploring...

christian-oreilly, 2023-02-20:
I did not look your notebook yet, but the only additional thing that would need to be integrated (if it is not already done) is to connect call back to addition/deletion of these shapes. These callbacks are necessary to update the annotations in the `inst` accordingly. If there is no problem triggering callbacks on these events, I see no technical issues in deploying that in the QC interface.

PR #82 (2023-04-06): Dash annotating
Closes #31
Closes #66 

Updated the annotating method in dashboard. This drops the dependency for `dash-extension` and does not require a keyboard press.

You can now modify and delete bad annotations in teh dashboard (before, you could only make new ones) (Enhancement).

TODOS:

- [x] Sync annotations across ICA and EEG visualizers. (need to redo this after my refactor).
- [x] Codespell Flake & docstring
closed 2023-04-06 by scott-huberty