lina-usc/pylossless #41: BUG: ica files not saving with '_Ica" suffix in derivatives folder

scott-huberty, 2023-03-07:
Given line 1231, it looks like the ICA files should save with the suffix '_ica' but they are not (I just tested saving the pipeline on my local machine).  

https://github.com/lina-usc/pylossless/blob/f9a2e2f443425b2c21d5110f178692a5d76c0383/pylossless/pipeline.py#L1223-L1255

Until we fix this, the QCR dashboard will return an error when trying to load the ICA

christian-oreilly, 2023-03-07:
Well, first, the comments in that block are off (it is loading, not saving... they are obviously from a copy-and-past from the corresponding function).

christian-oreilly, 2023-03-07:
This line:
https://github.com/lina-usc/pylossless/blob/f9a2e2f443425b2c21d5110f178692a5d76c0383/pylossless/pipeline.py#L1089
should have a `+ "_ica"` to fix this issue.

scott-huberty, 2023-03-07:
Yup -that confused me because wasn't even the block in question. The fix needed to be here:

https://github.com/lina-usc/pylossless/blob/f9a2e2f443425b2c21d5110f178692a5d76c0383/pylossless/pipeline.py#L1066-L1091

scott-huberty, 2023-03-07:
> This line:
> 
> https://github.com/lina-usc/pylossless/blob/f9a2e2f443425b2c21d5110f178692a5d76c0383/pylossless/pipeline.py#L1089
> 
> 
> should have a `+ "_ica"` to fix this issue.

See my PR

scott-huberty, 2023-03-07:
> Well, first, the comments in that block are off (it is loading, not saving... they are obviously from a copy-and-past from the corresponding function).

Can you request a change in your PR review to change that?

christian-oreilly, 2023-03-07:
Yeah, I saw this after. You change is good.

closed 2023-03-07 by scott-huberty

christian-oreilly, 2023-03-07:
> > Well, first, the comments in that block are off (it is loading, not saving... they are obviously from a copy-and-past from the corresponding function).
> 
> Can you request a change in your PR review to change that?

Just made the edits on PR #42

PR #42 (2023-03-07): FIX: saving ICA with correct suffix
closes #41 

Turns out that this was not an issue of diverging branches. 

Again, the bug was caused by us during development. Another reminder that we are at the stage of needing CI and tests.