lina-usc/pylossless #60: QC dashboard on offline machines

scott-huberty, 2023-03-14:
I just wanted to leave a note here for discussion, because I imagine it will come up at some point, particularly if @andesha decides to implement pylossless on the clusters.

Since we are using `dash-bootstrap-components`, our CSS style sheets are being pulled from a server, and thus requires an internet connection.

There is a way to download and store/serve these locally (for example in an `assets` folder) for cases where the machine doesn't have internet access, but I've yet to look into this.

christian-oreilly, 2023-03-16:
Yes, agreed, internet access should definitely be optional to run the pipeline and the QC...

Andesha, 2023-03-27:
As you'll need the internet at least one to install things (:laughing:) we can maybe just package in a small shell script that does a `curl` or `wget`? :thinking:

scott-huberty, 2023-04-05:
> we can maybe just package in a small shell script that does a `curl` or `wget`? 🤔

yes I think this is a great idea!