lina-usc/pylossless #64: Handling Multi-modal recordings

scott-huberty, 2023-03-16:
if I run this

```
import pylossless as ll 
import mne
fname = mne.datasets.sample.data_path() / 'MEG' / 'sample' /  'sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname, preload=True).pick_types(eeg=True)  # Without explicit exclusion of MEG chans the pipeline breaks. Must fix.

config = ll.config.Config()
config.load_default()
config.save("my_project_ll_config.yaml")

pipeline = ll.LosslessPipeline('my_project_ll_config.yaml')
pipeline.run_with_raw(raw)
```

If I don't explicitly pick `eeg` channels like in line 4, the pipeline will flag all channels in `flag_ch_outliers` and fail.
This is especially problematic for users who run the pipeline from a BIDS path, and don't have the option to pick `eeg` chans.

christian-oreilly, 2023-03-16:
I'd suggest adding something to the config file to define which channels types should be included, maybe with a default behavior to "EEG" and "MEG? We should definitely not have to pick_types explicitly like that for basic usage.

christian-oreilly, 2023-04-29:
Upgraded from bug to enhancement. The bug was fixed by PR #92, which explicitly operates only on EEG. The enhancement would require, e.g., specifying modalities (EEG, MEG, ...) to be included as part of the config file.

closed 2024-11-27 by christian-oreilly