lina-usc/pylossless #66: Enforce flak8, codespell, pydocstyle in pylossless.dash module

scott-huberty, 2023-03-16:
Next up on our journey towards PEP8 / Numpydoc compliance.

These ones might be a doozy.

Andesha, 2023-03-20:
I'll likely get to this soon, so don't start it yourself :wink:

scott-huberty, 2023-03-20:
Have fun!!!! 😉

Andesha, 2023-03-27:
What's the codespell thing you use?

scott-huberty, 2023-03-27:
@Andesha let me know if you want me to merge #62 now or after your done.

There is a decent chance that there will be some conflicts that need to be resolved if you've already started the linting work. 

I am fine with handling conflicts in #62 if needed.

scott-huberty, 2023-03-27:
Ok i'll wait to merge it until we have your PR merged.

> "too many instance variables"

interesting... never heard of this one.

scott-huberty, 2023-03-27:
@Andesha what linter are you using?

Andesha, 2023-03-27:
Odd... now that you mention it I was trying to use flake8... :thinking:

Andesha, 2023-03-27:
good thing you noticed that - I had conflicting linting tools going and it was causing huge issues...

its been fixed now and is also showing way less problems so I should be done shortly(-ish)

Andesha, 2023-03-28:
can prolly close this now due to #76

PR #82 (2023-04-06): Dash annotating
Closes #31
Closes #66 

Updated the annotating method in dashboard. This drops the dependency for `dash-extension` and does not require a keyboard press.

You can now modify and delete bad annotations in teh dashboard (before, you could only make new ones) (Enhancement).

TODOS:

- [x] Sync annotations across ICA and EEG visualizers. (need to redo this after my refactor).
- [x] Codespell Flake & docstring
closed 2023-04-06 by scott-huberty