lina-usc/pylossless #73: Add Artifact Blocking as an optional preprocessing step

christian-oreilly, 2023-03-19:
Artifact blocking is described here: https://www.researchgate.net/publication/4248725_A_Simple_and_Fast_Algorithm_for_Automatic_Suppression_of_High-Amplitude_Artifacts_in_EEG_Data

This approach has been suggested has being advantageous compared to ICA for infants data.  We propose it here as an omnibus artifact rejection optional step, not a replacement to ICA because beside rejecting artifact, ICA is also used to classify them.

christian-oreilly, 2023-03-19:
@dsrishyla FYI, here is the Python translation of the Matlab code you obtained from the authors: https://colab.research.google.com/drive/1LM_fkuxR2XiQRmT6kVupkcRlAHhZ1AsU

Andesha, 2023-03-20:
Just a note - any sort of extra preprocessing can be called as part of the staging script

does that have support currently?

Andesha, 2023-03-20:
Assuming you have perfect knowledge of what goes on in recordings (which never happens lol) you would do that sort of recording gap thing inside of the init procedure.

Staging is meant to be your "last chance before we actually interpret this signal" step. i.e., you find out that in a particular subject you NEED to manually mark out a channel but don't want to re-init it.

A second example would be wanting to mark more time as out of task as sort of outlined above. You could re-init (which sucks) or put some stuff into staging.

Andesha, 2023-03-20:
That sounds great. Also as long as the docs mention something along the lines of "be very sure about your staging scripts, and don't run other people's without checking first" you're probably fine...

christian-oreilly, 2023-03-20:
Yeah, but using a callback like this might somewhat by-pass the issue altogether since you can just have this function defined at the top of the script where you run the pipeline... or you import it from another file like any other functions, so I guess it can be argued not be more dangerous than any other import... although I am not a security guy :)

scott-huberty, 2023-04-17:
I'm +1 for adding this, preferable for v.02.

Based on my limited knowledge of AB, if my assumption is right that it can be applied to low-density EEG ( 32 or less channels), which ICA is said to not be so good for - I think this could be a good selling point for the pipeline and it would open it up to users with low-density systems.

I also think this could be a nice opportunity for Diksha to become involved!

christian-oreilly, 2024-12-09:
Adding the code for Artifact Blocking is simple. However, we might want to think about how we want to include this step. Should we have a mechanism to add `preprocessing steps`? If so, how flexible do we want to be about it? We can have them all run at a specific point (e.g., at the beginning, or just before ICA). Or do we want to have the flexibility to specify at what stage of the pipeline this step is run? Going one step deeper would be to actually create a pipeline where the user can "register" whatever steps they want in whatever order they want, with the default being the current steps (the EEG-IP-L steps). That would be a significant refactoring that would come in a later milestone.