lina-usc/pylossless #79: Lossless Logging Localization

Andesha, 2023-03-28:
In the old version there was a fairly robust logging system for each stage that was being executed.

Previously I believe this was just information being printed to `stdout` and then redirected to a file through a SLURM option...

What we should do instead is some of the following:

- Figure out what from the MNE side of things is being written to `stdout` and find out if we need it for logging
  - We can capture this somehow I think
- Agree on the location of these files
  - I propose the same as before at `derivatives/lossless/logs` type folder
- Create the name of the file based on `$FILENAME_$TIMESTAMP.ll_log`
- Prepend the file with the contents of the config being used to run it, just in case
- As a first step, wrap the various function calls inside of `_run` in `pipeline.py` with basic logging calls like
  - "Setting Montage", "Finding Breaks", "Flagging comically bad outliers", etc
- Add some timing information in these steps as well for profiling on clusters in the future

scott-huberty, 2023-03-28:
Bonus points if we can include some emojis like :white_check_mark: for major steps of the pipeline ; )

Andesha, 2023-03-28:
> Bonus points if we can include some emojis like white_check_mark for major steps of the pipeline ; )

I don't know what that would do on the cluster, but let's try :laughing: 

> We _could_ just use `mne.utils.logger` in our pipeline too (i.e. `logger.info()`, `logger.debug()`. ) you'll see in `pylossless.pipeline.py` that I already imported this logger.

I had forgotten (or not noticed?), but that sounds just right!

closed 2023-04-29 by scott-huberty