lina-usc/pylossless #86: Adding diagnostic functions

christian-oreilly, 2023-04-04:
We should add functions that allow, given a config file and a raw object, to generate diagnostic figures for each of the steps. Representations shown in the first two figures of this paper https://pubmed.ncbi.nlm.nih.gov/33038417/ are probably a good starting point.

scott-huberty, 2023-04-05:
It actually might be helpful for us to have some form of this done by paper submission (so end of June)?

christian-oreilly, 2023-04-06:
Sure. But I think this is still milestone 0.02 (i.e., not blocking 0.01).

christian-oreilly, 2025-02-03:
I think we could add functions for plotting figures like the panels A and B in this figure:

[image]

It could be useful also to have a plot of flagged topomaps and a plot of the valid topomaps. 

Part of this panel could also be useful for diagnosis:

[image]