lina-usc/pylossless #90: Topomap refactoring

christian-oreilly, 2023-04-07:
The objective of this issue is to refactor the topomap code so that the generation of a single topomap is a function or class of its own, so that it can be more easily reused outside the specific context of PyLossless (for an eventual fork of the visual components to their own project).

scott-huberty, 2023-04-07:
> The objective of this issue is to refactor the topomap code so that the generation of a single topomap is a function or class of its own, so that it can be more easily reused outside the specific context of PyLossless (for an eventual fork of the visual components to their own project).

I've had this on my ind as well.. +1!

christian-oreilly, 2023-04-07:
Yeah... that was self-interested. I want to make a topoplot now for some project and instead of going the usual way of tweaking mne code to get it done, I thought I give this a shot.

scott-huberty, 2023-04-07:
Well.. If you end up getting to it on your own - let me know if any of the topo Docstring is incorrect/confusing - I did my best to represent the code but i'm sure I made mistakes.

scott-huberty, 2023-04-07:
> that was self-interested. I want to make a topoplot now for some project and instead of going the usual way of tweaking mne code to get it done, I thought I give this a shot.

Nah IMO this is a perfect example of how a feature request should be done.. Driven by a real and reasonable use-case!

christian-oreilly, 2023-04-09:
That thing is mostly done. Took much more time that I cared for, but it is done. For plotting a single topomap:

[image]

For a grid of topomaps:

[image]

For embedding it as a dash component of an existing app:

[image]

For an on-the-fly standalone dash app:

[image]

These classes are properly integrated, such that you can get a plotly figure from the dash objects:

[image]

@scott-huberty We should discuss package naming because I think it would be a good occasion to push this into a new repo in the context of #94 (which will also support #33).

scott-huberty, 2023-04-10:
Looks awesome! 

Yes, let's chat this week

PR #102 (2023-04-28): 90 topomap refactoring
Fix #90
closed 2023-04-28 by christian-oreilly