lina-usc/pylossless #93: The colab example is not accessible anymore

christian-oreilly, 2023-04-09:
In previous versions, the colab notebook example was accessible through the readme of the GitHub repo. With the previous refactoring, this bit has been removed from the readme without being added back anywhere else in the doc.

scott-huberty, 2023-04-09:
> In previous versions, the colab notebook example was accessible through the readme of the GitHub repo. With the previous refactoring, this bit has been removed from the readme without being added back anywhere else in the doc. 

Feel free to put it in tbe README.

christian-oreilly, 2023-04-10:
Yes. That is a possibility. But we may want to consider alternatives before I do so (e.g., having a section of our doc for notebooks). I am not so sure about it.

scott-huberty, 2023-04-10:
> having a section of our doc for notebooks

Sure, definitely.

FYI I did link to the notebook [here](https://pylossless.readthedocs.io/en/latest/generated/auto_tutorials/usage.html#sphx-glr-generated-auto-tutorials-usage-py). 

But I need to rename that notebook - I haven't had the time to substantially work on the doc lately. I'll get to it after we finish the development for .01

christian-oreilly, 2023-04-10:
> FYI I did link to the notebook [here](https://pylossless.readthedocs.io/en/latest/generated/auto_tutorials/usage.html#sphx-glr-generated-auto-tutorials-usage-py).

This is not the Colab example. I won't run on the Colab.

scott-huberty, 2023-04-10:
> > FYI I did link to the notebook [here](https://pylossless.readthedocs.io/en/latest/generated/auto_tutorials/usage.html#sphx-glr-generated-auto-tutorials-usage-py).
> 
> 
> 
> This is not the Colab example. I won't run on the Colab.

There is a link to the collab example.

christian-oreilly, 2023-04-12:
The link is available. The discussion on a section for notebooks is not pressing, so I scheduled it for the 0.02 release.

closed 2023-05-08 by scott-huberty