lina-usc/pylossless #96: Re-referencing is causing weird signal properties?

scott-huberty, 2023-04-11:
I _assume_ this is only relevant to my development branch where we moved `flag_outlier_chs` into `rereference` (but I can confirm when I switch to the main branch.

BEFORE running the pipeline:

[image]

Looks good.

After running `flag_ch_sd_ch` (the first step of the pipeline. The recording is re-referenced 2 times by the end of this step FYI...)

[image]

the signals are not legible.

scott-huberty, 2023-04-11:
The trick seems to be to do a `pick_types(eeg=True)` here:

https://github.com/lina-usc/pylossless/blob/2095f13eff76c5abd7729340642b57424511a207/pylossless/pipeline.py#L81-L83

[image]

Which blows my mind a little.. I'd assume that by default MNE wouldn't let a non-data channel (i.e. STIM channel) be a reference!

christian-oreilly, 2023-04-11:
> Which blows my mind a little.. I'd assume that by default MNE wouldn't let a non-data channel (i.e. STIM channel) be a reference!

I think it probably does use only EEG channels by default but that it assumes that we know what we are doing so if we give it explicitly a list of channels, it uses that list without doing further exclusion... and I suspect that we were explicitly passing this stim channel to the rereferencing function:
```python
        inst.set_eeg_reference(ref_channels=[ch for ch in inst.ch_names
                                             if ch not in self['manual']],
                               **kwargs)
```

PR #92 (2023-04-27): ENH, FIX: Moved flag_outlier_chs and added fixed_threshold method.
Closes #55
Closes #56 
Closes #88 
Closes #91 
Closes #95 
Closes #96  

Work from dev session on april 7th. 

Maybe wait to merge this until after the (hopeful) pymat dev session on tuesday the 11th

NOTE, STILL TODO:

- [x] update `pytest`  to reflect updated pipeline methods
closed 2023-04-27 by scott-huberty