lina-usc/pylossless #99: Find Breaks Workflow Requires Annotation Structure in Raw

Andesha, 2023-04-18:
When passing `find_breaks` to the pipeline like the following:

```
config['find_breaks'] = {}
config['find_breaks']['min_break_duration'] = 10
config['find_breaks']['t_start_after_previous'] = 3
config['find_breaks']['t_stop_before_next'] = 0
config['find_breaks']['ignore'] = ['boundary', 'e-255']
```

The pipeline crashes if there is not an annotation structure already created in the data. Error here:

```
  File "/scratch/tk11br/TRI-pipeline/env/lib/python3.10/site-packages/mne/preprocessing/artifact_detection.py", line 471, in annotate_break
    raise ValueError('Could not find (or generate) any annotations in '
ValueError: Could not find (or generate) any annotations in your data.
```

I fixed this personally by doing this:

```
annos = mne.annotations_from_events(events=mne.find_events(raw, initial_event=True),
                                    sfreq=raw.info['sfreq'],
                                    orig_time=raw.info['meas_date'])
raw.set_annotations(annos)
```

Likely needs to be fixed before the initial release.

scott-huberty, 2023-04-19:
I see. 

You should also be able to pass config['find_breaks'] = `False` to skip this step.

I made the `find_breaks` behavior True by default because when I used MATLAB Lossless, finding breaks was not done by default, and this is something I always needed to do (but often forgot) which resulted in having to re-run the pipeline.

The default for `annotate_break` is to look for annotations, alternatively, you can pass in an event array (or set your annotations from your stim channel as you did). 

But I agree that we might want to handle a situation like yours more gracefully.

christian-oreilly, 2024-11-13:
I am not sure I get the problem. I tried the following:

```python

from pathlib import Path
import shutil

import pylossless as ll
from pylossless.datasets import load_openneuro_bids

raw, config, bids_path = load_openneuro_bids()

config["find_breaks"] = {}
config["find_breaks"]["min_break_duration"] = 9
config["find_breaks"]["t_start_after_previous"] = 1
config["find_breaks"]["t_stop_before_next"] = 0
config["ica"]["ica_args"]["run1"]["max_iter"] = 5000
config['find_breaks']['ignore'] = ['boundary', 'e-255']

pipeline = ll.LosslessPipeline(config=config)

not_in_1020 = ["EXG1", "EXG2", "EXG3", "EXG4", "EXG5", "EXG6", "EXG7", "EXG8"]
pipeline.raw = raw.pick("eeg", exclude=not_in_1020).load_data()
pipeline.run_with_raw(pipeline.raw)

shutil.rmtree(bids_path.root)
```

Has this issue already been solved by another PR or is that code not properly getting the issue?

scott-huberty, 2024-11-13:
@christian-oreilly in your code if you explicitly make sure there are no annotations in the `raw` object by doing:

```python
pipeline.raw = raw.pick("eeg", exclude=not_in_1020).load_data().set_annotations(None)
```

you'll get the error below, which actually emanates from MNE.

We could try to get MNE to add a parameter for emitting a `warning` instead of an error but it's probably easier for us just to implement a little work around in our code.

```python

ValueError                                Traceback (most recent call last)
Cell In[4], line 1
----> 1 pipeline.run_with_raw(pipeline.raw)

File ~/devel/repos/pylossless/pylossless/pipeline.py:1156, in LosslessPipeline.run_with_raw(self, raw)
   1154 """Execute pipeline on a raw object."""
   1155 self.raw = raw
-> 1156 self._run()
   1157 return self.raw

File ~/devel/repos/pylossless/pylossless/_logging.py:63, in lossless_time.<locals>.wrapper(*args, **kwargs)
     61 logger.info(" ⏩ LOSSLESS: Starting Pylossless Pipeline.")
     62 start_time = time.time()
---> 63 result = func(*args, **kwargs)
     64 end_time = time.time()
     65 dur = f"{(end_time - start_time) / 60:.2f}"

File ~/devel/repos/pylossless/pylossless/pipeline.py:1170, in LosslessPipeline._run(self)
   1167 self.run_staging_script()
   1169 # find breaks
-> 1170 self.find_breaks(message="Looking for break periods between tasks")
   1172 # OPTIONAL: Flag chs/epochs based off fixed std threshold of time axis
   1173 self.flag_epochs_fixed_threshold()

File ~/devel/repos/pylossless/pylossless/_logging.py:47, in lossless_logger.<locals>.wrapper(message, *args, **kwargs)
     45 elif verbose:
     46     logger.info(f"LOSSLESS: 👇 {this_step}.")
---> 47 result = func(*args, **kwargs)
     48 end_time = time.time()
     49 dur = f"{end_time - start_time:.2f}"

File ~/devel/repos/pylossless/pylossless/pipeline.py:709, in LosslessPipeline.find_breaks(self)
    707 if "find_breaks" not in self.config or not self.config["find_breaks"]:
    708     return
--> 709 breaks = annotate_break(self.raw, **self.config["find_breaks"])
    710 self.raw.set_annotations(breaks + self.raw.annotations)

File <decorator-gen-205>:12, in annotate_break(raw, events, min_break_duration, t_start_after_previous, t_stop_before_next, ignore, verbose)

File ~/miniforge3/envs/lossless/lib/python3.10/site-packages/mne/preprocessing/artifact_detection.py:543, in annotate_break(raw, events, min_break_duration, t_start_after_previous, t_stop_before_next, ignore, verbose)
    538     annotations = annotations_from_events(
    539         events=events, sfreq=raw.info["sfreq"], orig_time=raw.info["meas_date"]
    540     )
    542 if not annotations:
--> 543     raise ValueError("Could not find (or generate) any annotations in your data.")
    545 # Only keep annotations of interest and extract annotated time periods
    546 # Ignore case
    547 ignore = tuple(i.lower() for i in ignore)

ValueError: Could not find (or generate) any annotations in your data.
```

scott-huberty, 2024-11-13:
Just submitted a fix and requested your review @christian-oreilly