lina-usc/pylossless #111: Re-implement MATLOSSLESS ICA replicability assessment

scott-huberty, 2023-05-03:
As discussed in slack:

@jimadesjardins :

" _When designing something like the lossless pipeline in the context of topics like ICA minimum requirements I tend to "give up".. and rather implement a test to determine if the data were in fact sufficient... this is why the lossless pipeline runs 3 final ICA decompositions on the flagged data... if the ICA replicates across the 3 ICA decompositions the data were sufficient... if they don't the data are not sufficient._ "

and

" _My belief is that all papers that use ICA should be reporting whether or not their ICA decomposition replicates... one of my dreams is that the lossless pipeline would make that feasible._ "

@christian-oreilly: 

" _I see no reason not to implement the 3 ICA run now that we are in good shape… whether or not it is run would be controlled by the config file … I don’t think it would be limited to HPC users… nowaday,  many personal laptop have many cores…. computational load is less and less of a problem as time passes…_ "

So maybe we can re-implement the the 2 additional ICA runs and the replication assessment between the final ICA's. 
I'm +1 for making this optional, with the default set to _not_ running 2 additional ICA's....

scott-huberty, 2023-05-03:
Also, as I remember @jadesjardins @Andesha , replication "annotating" really happened on a epoch by epoch basis? As in I remember seeing time periods annotated with no replication, but not entire decompositions. Is that correct?

Andesha, 2023-05-05:
Not sure what you mean :thinking:

scott-huberty, 2023-05-05:
> Not sure what you mean 🤔

I could be wrong but I'm pretty sure In the QC interface I remember seeing periods of time where the IC traces opacity was low (like 0.5) - I think I remember @jadesjardins saying this meant that the IC time-series did not replicate during that period of time.

Does that sound familiar?

Andesha, 2023-05-05:
oh - that's the log likelihood. That's where AMICA is saying those periods in time did not contribute well to its decomposition and is considering it less.

scott-huberty, 2023-05-05:
> oh - that's the log likelihood. That's where AMICA is saying those periods in time did not contribute well to its decomposition and is considering it less.

Ohhhh okay. 

So can you clarify when you said that Lossless flags non-replication, what did that look like in MATLAB? Should we have a `replicated` column in our `flagged_tsv` that has `True` or `False`?

scott-huberty, 2023-05-05:
> We can't do that without the multiple ICA runs during the same step
> 
> [image]
> 
> The `ic_rt` flagged components there with the blue triangles are components that did not replicate across the 3 ICAs during the 4th step

Got it, thanks! Maybe we can add this info to each components `Hoverlabel`