lina-usc/pylossless #119: `EEGAnnotationList` not JSON serializable

scott-huberty, 2023-05-08:
There is a silent failure when serving the QC dashboard:

```
  File "/usr/local/Cellar/python@3.11/3.11.3/Frameworks/Python.framework/Versions/3.11/lib/python3.11/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type EEGAnnotationList is not JSON serializable
```

I think `EEGAnnotationList` is a pandas series where it should be a dictionary (see this [thread](https://stackoverflow.com/a/71627993/16745182)).

I spent the weekend in a battle to deploy the dashboard to a PAAS service (Heroku/Google Cloud/Azure) to no avail. After exhausting all other avenues, I have a hunch this is the underlying issue. 

I expect that deploying to a PAAS service won't be possible until this is corrected.

christian-oreilly, 2023-05-09:
A full stack error would be more useful than just the last lines. Note that, in the end, we don't need the dcc.Store... this could just be replaced by a plain old dictionary I think.

christian-oreilly, 2024-12-10:
It is currently unclear how to reproduce this issue. In #199, I created a code to test `EEGAnnotationList`, but I'd like to understand when and how the serialization gets triggered in the operation of `Dash` to create this issue. It is currently unclear, so it is unclear what is the best path to solve this issue.

christian-oreilly, 2024-12-10:
Maybe implementing #46 could give us an opportunity to trigger that bug. Before doing this, we need to fix #185 to allow running tests using our test data.

christian-oreilly, 2024-12-13:
I suggest to close this with PR #199. This PR makes a fair attempt at triggering this issue, but without more information, I cannot reproduce. The `EEGAnnotationsList` is not JSON serializable as is and making it intrinsically serializable is probably not the way to go. It is more common to add for example a `to_json` method which returns a JSON string for the object. However, implementing such an approach requires being able to trigger the issues, and testing what circumstances make the software crash. I propose to close this issue with PR #199 and reopen if/when we can reproduce it. So far, Dash tests do not trigger this issue.

scott-huberty, 2024-12-13:
And yes feel free to close.

christian-oreilly, 2024-12-13:
> Sorry - It's been too long for me to remember how to trigger it (and thanks for taking the time for trying to do so)

No problem. I figured that it was probably the case. Hence why I proposed to close at this point.

christian-oreilly, 2025-01-28:
In IPython, running the content of app.py

```python
import dash
import dash_bootstrap_components as dbc
from pylossless.dash.qcgui import QCGUI

def get_app(fpath=None, project_root=None, disable_buttons=False, kind="dash"):
    """Call either Dash or Jupyter for Lossless QC procedure."""
    if kind == "jupyter":
        from jupyter_dash import JupyterDash

        app = JupyterDash(__name__, external_stylesheets=[dbc.themes.SLATE])
    else:
        app = dash.Dash(__name__, external_stylesheets=[dbc.themes.SLATE])

    QCGUI(app, fpath=fpath, project_root=project_root, disable_buttons=disable_buttons)
    return app

get_app().run_server(debug=False, use_reloader=False)
```

I got

```
Dash is running on http://127.0.0.1:8050/

 * Serving Flask app '__main__'
 * Debug mode: off
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on http://127.0.0.1:8050
Press CTRL+C to quit
127.0.0.1 - - [28/Jan/2025 12:48:40] "GET /_reload-hash HTTP/1.1" 200 -
127.0.0.1 - - [28/Jan/2025 12:48:40] "GET /_dash-dependencies HTTP/1.1" 200 -
Exception on /_dash-layout [GET]
Traceback (most recent call last):
  File "/Users/christian/opt/anaconda3/lib/python3.9/site-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/Users/christian/opt/anaconda3/lib/python3.9/site-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/Users/christian/opt/anaconda3/lib/python3.9/site-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/Users/christian/opt/anaconda3/lib/python3.9/site-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/Users/christian/opt/anaconda3/lib/python3.9/site-packages/dash/dash.py", line 659, in serve_layout
    to_json(layout),
  File "/Users/christian/opt/anaconda3/lib/python3.9/site-packages/dash/_utils.py", line 22, in to_json
    return to_json_plotly(value)
  File "/Users/christian/opt/anaconda3/lib/python3.9/site-packages/plotly/io/_json.py", line 123, in to_json_plotly
    return json.dumps(plotly_object, cls=PlotlyJSONEncoder, **opts)
  File "/Users/christian/opt/anaconda3/lib/python3.9/json/__init__.py", line 234, in dumps
    return cls(
  File "/Users/christian/opt/anaconda3/lib/python3.9/site-packages/_plotly_utils/utils.py", line 59, in encode
    encoded_o = super(PlotlyJSONEncoder, self).encode(o)
  File "/Users/christian/opt/anaconda3/lib/python3.9/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/Users/christian/opt/anaconda3/lib/python3.9/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/Users/christian/opt/anaconda3/lib/python3.9/site-packages/_plotly_utils/utils.py", line 136, in default
    return _json.JSONEncoder.default(self, obj)
  File "/Users/christian/opt/anaconda3/lib/python3.9/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type EEGAnnotationList is not JSON serializable
```

The served page display only "Error loading layout" probably because of this issue.

scott-huberty, 2025-01-28:
aha this is it. 

EEGAnnotaitonList is a Pandas DataFrame - Seems like there's an issue sending this from the server to client (which requires JSONifying the data?)

christian-oreilly, 2025-01-28:
It is Series, not a DataFrame, but I get the point :)

I just replaced the Series by and `OrderedDict` and it is still not serializable. I think the class contains Dash objects and I suspect that these are not all serializable

christian-oreilly, 2025-01-28:
Confirmed. Problem fixed. This issue will be closed with the PR associated in issue #209

PR #199 (2024-12-13): Fix #119, #182, and #185
Closes #119
Closes #182
Closes #185
commit 233a5364 (2024-12-13): Fix #119, #182, and #185 (#199)

* Adding a test for reproducing #119
* Fix #182
* Add and fix some dash tests
PR #219 (2025-01-29): Fix 209 eeg signals appears flat in the dashboard
Fix #209 
Fix #119
closed 2025-01-29 by christian-oreilly