lina-usc/pylossless #121: Proposed API Changes:

scott-huberty, 2023-05-11:
I wanted to propose this while this is still an internal effort (i.e. outside users don't appear to be using this pakcage yet):

CC @Andesha and @jadesjardins to chime in:

| Current API          | Proposed API |
|------------------|--------------|
| `flag_ch_sd_ch` | `flag_noisy_channels` |
| `flag_ch_sd_epoch`  | `flag_noisy_epochs`  |
| `flag_ch_low_r`  | `flag_uncorrelated_channels`  |
| `flag_epoch_low_r` | `flag_uncorrelated_epochs` |
| `flag_ch_bridged` | `flag_bridged_channels` |
| `flag_ch_rank` | `flag_rank_channel` |

The new naming convention aims to be more pythonic and is inspired by MNE / pyprep - For example you have `mne.preprocessing.compute_bridged_channels` not `compute_some_metric_abbreviated`

Outside users are not going to be familiar with our internals and so something like `ch_sd_ch` is vague. I think it is better to communicate _conceptually_ what the function does.

Andesha, 2023-05-15:
Big fan of this as long as the reference documentation mentions the old names ;)

Andesha, 2023-05-29:
I've given it some more thought - I don't really like "noisy" as  name for ch_sd...

scott-huberty, 2023-05-29:
we shouldn't use "uncorrelated" again because we need to differentiate between the flags of the different steps. 
"bad" is probably not a good alternative bc (at least in `mne`) "bad" is a special term.

The function basically calculates the std. across the samples of each epoch. The heuristic is, if the std is high, it's indicative of the channel being what we'd colloquially call "noisy" during that epoch. So for me "noisy" felt like a natural label.

you maybe could say the same about the `flag_uncorrelated` step but to me this is a more fine grained/low-amplitude artifact and that's why it is not picked up by the "noisy" step.

PR #152 (2023-11-07): Rename flags and methods
Closes #121 
Closes #122

To be in line with our paper:

| Current  | to | New |
| ------------- |--- |------------- |
| `flag_ch_sd_ch`| ➡️ |  `flag_noisy_channels` & `flags["ch"]["noisy"]`  |
| `flag_ch_low_r`  | ➡️ | `flag_uncorrelated_channels` & `flags["ch"]["uncorrelated"]`  |
| `flag_ch_bridge` | ➡️ |   `flag_bridged_channesl` & `flags["ch"]["bridged"]` |
| `flag_ch_rank`  | ➡️ |  `flag_rank_channel` & `flags["ch"]["rank"]`
| `flag_ch_sd_epoch` | ➡️ |  `flag_noisy_epochs` |
| `flag_low_r_epochs` | ➡️ |  `flag_uncorrelated_epochs` |
| `flag_ic_sd_1` | ➡️ | `flag_ [... cut]
closed 2023-11-07 by scott-huberty