lina-usc/pylossless #122: Decide on API

scott-huberty, 2023-05-11:
While API is on my mind - We should make a decision on what we want our public API to be and make everything else private.

This will help organize the package and will make it clear to us which functions/methods are public and thus should be stable. It also relieves us of the work of needing to document/maintain functions that are only used internally.

CC @Andesha

christian-oreilly, 2023-05-11:
It is unlikely that we will change a function from private to public. Maybe just listing the public functions that you think should be made private and we can discuss these, specifically? I am not a big fan of trying to put everything private (not saying that it is what you aim to do)... I'd avoid putting private anything that might have an external use (even though we may not have coded it with this in mind) and putting private only what is very specific to the internals and unlikely to ever be useful for re-use... I am not rigid on that position, but that is how I see it at this point. This is a research tool, aim at researchers who are expected (and should feel welcome) to play with the code beyond the narrowly drawn borders of the use case we may have thought about...

scott-huberty, 2023-05-11:
Sure, I can propose a list of functions that should not be part of the API.

I agree with you that there are some utilities outside the main pipeline methods that can be useful and be a part of the API.

PR #152 (2023-11-07): Rename flags and methods
Closes #121 
Closes #122

To be in line with our paper:

| Current  | to | New |
| ------------- |--- |------------- |
| `flag_ch_sd_ch`| ➡️ |  `flag_noisy_channels` & `flags["ch"]["noisy"]`  |
| `flag_ch_low_r`  | ➡️ | `flag_uncorrelated_channels` & `flags["ch"]["uncorrelated"]`  |
| `flag_ch_bridge` | ➡️ |   `flag_bridged_channesl` & `flags["ch"]["bridged"]` |
| `flag_ch_rank`  | ➡️ |  `flag_rank_channel` & `flags["ch"]["rank"]`
| `flag_ch_sd_epoch` | ➡️ |  `flag_noisy_epochs` |
| `flag_low_r_epochs` | ➡️ |  `flag_uncorrelated_epochs` |
| `flag_ic_sd_1` | ➡️ | `flag_ [... cut]
closed 2023-11-07 by scott-huberty