lina-usc/pylossless #131: Fix the montage rereference

christian-oreilly, 2023-06-06:
Currently the `pipeline` function `wrap_locs` appears not be used anywhere. Similarly, `self.config['replace_string']['montage_info']` is used only in `wrap_locs`. I also think we did not implemented the referencing on a common surface. We need to fix these things.

scott-huberty, 2023-06-06:
`self.config['replace_string']` definitely sounds like a hold over from the MATLAB config, If I recall correctly. 

> I also think we did not implemented the referencing on a common surface.

True. But we never decided on whether it's worth the effort?

christian-oreilly, 2023-06-07:
> True. But we never decided on whether it's worth the effort?

Indeed. That is the point... we should decide and act on it (i.e., remove wrap_locs if not, implement it if yes). Following the previous exchange with @jadesjardins I tend to lean on the side that it is relevant to do something... particularly for the pooling of dataset recording with different systems and having a different coverage of the head (e.g., EGI including the lower back of the head vs a 10-20 montage that did not) since this coverage difference can bias the reference. 

> self.config['replace_string'] definitely sounds like a hold over from the MATLAB config, If I recall correctly.

Yes, but I am not so much reference to the 'replace_string' of that code but the 'montage_info' and the fact that we currently have two place where the montage can be specified.

scott-huberty, 2023-06-07:
> Indeed. That is the point... we should decide and act on it (i.e., remove wrap_locs if not, implement it if yes). Following the previous exchange with @jadesjardins I tend to lean on the side that it is relevant to do something... 

Agreed. Now we just have to find someone to do the job ;)

scott-huberty, 2023-06-26:
@Andesha I haven't looked at dyour comment/PR Yet but yeah the co-registration to the 10-20 surface (to be used for robust average referencing) is something we want to implement but it's a bit complex so never finished it.

If you look at the `warp_locs` code you'll see what mne function is called, and you can check the imports to see what module it is coming from. 

The MNE site will definitely have an API documentation and probably a tutorial for it.

Ping us with any questions, we're happy to help.