lina-usc/pylossless #134: Re-referencing concerns

Andesha, 2023-06-08:
This is just a concern, but I've been working a lot more handson with data coming from the pipeline and am not sure that after we are rejecting a source (channels) that we are rereferencing the data.

I see that there's a function for it for channels [here](https://github.com/lina-usc/pylossless/blob/b5079832d5283981e5f209dd54fe2e52fd42137d/pylossless/flagging.py#L83) but it does not appear to be used after things like the rank channel are thrown out...

Andesha, 2023-06-08:
Poking through a bit more of the code it seems like it's done via the `get_epochs` function - can someone just confirm this is being done correctly?

scott-huberty, 2023-06-09:
I accidentally deleted my response. I'll post it again:

robust average `re-referencing` happens anytime `get_epochs` is called:

https://github.com/lina-usc/pylossless/blob/b5079832d5283981e5f209dd54fe2e52fd42137d/pylossless/pipeline.py#L520-L521

[Here](https://github.com/lina-usc/pylossless/blob/b5079832d5283981e5f209dd54fe2e52fd42137d/pylossless/pipeline.py#L556-L558):
```
if rereference:
    self.flags["ch"].rereference(epochs)

```

MOST pipeline steps will call `get_epochs` before starting, thus average re-referencing the data, excluding any channels that are in `pipeline.flags['ch']` or `pipeline.raw.info['bads']` from the average.

The exceptions are `flag_ch_bridge` and `flag_ch_rank`:

https://github.com/lina-usc/pylossless/blob/b5079832d5283981e5f209dd54fe2e52fd42137d/pylossless/pipeline.py#L1070-L1083

above, `flag_ch_low_r` calls `get_epochs`. At the end of it's funciton it returns the 2D matrix of nearest neighbour values that it operated on, which we call `data_r_ch`. 

`flag_ch_bridge` and `flag_ch_rank` operate on `data_r_ch` directly so they don't themselves call `get_epochs` and thus the data is NOT robust average re-referenced before.

The next time robust average referencing occurs is `flag_epoch_low_r`, which calls `get_epochs`. 

@Andesha @jadesjardins if this deviates from the MATLAB code, let us know.

scott-huberty, 2023-06-14:
Thanks @jadesjardins !

From what I can tell, [MATLAB lossless didn't re-reference](https://github.com/BUCANL/EEG-IP-L/blob/a56520698ec87bca2fd8897ac248454e03a35add/code/scripts/s01_scalpart.htb#L271-L358
) after `flag_ch_low_r`,  or after`flag_ch_bridged` , _either_ : 

Looking at the MATLAB code and my [description of pylossless above](https://github.com/lina-usc/pylossless/issues/134#issuecomment-1583787531), it seems like we directly translated the MATLAB code.

Maybe this was a bug in MAT Lossless that we ported over to Python. But when I think about it, we exclude channels from the average so that noisy channels don't skew the ref. `Bridged` channels don't quite fit that bill I don't think, so I'm not sure how much of an impact adding another re ref would have. I kind of feel the same about `low_r` chans.

Unless you feel strongly that skipping the re-ref was a bug that pyLossless should correct (or unless I'm reading the matlab code incorrectly) I'm tempted to consider this a _feature request_ and _not_ a __bug_ fix._

In that case I'm definitely +1 to add another re-reference if others think it's important, but personally I won't have time to do it I don't think. but if anyone is really pressing for this step and implements it in their local fork, a PR is welcome

scott-huberty, 2023-06-14:
> The rank channel remains in the scalp channel data, but this channel is not included in the ICA decomposition.

We are excluding `rank_channel` from following pipeline steps such as `flag_epoch_low_r`, so I think I'd consider this part a bug and something we should fix.

Andesha, 2023-06-14:
I'm pretty sure it does the rereference in the snippet you linked here:

[image]

Andesha, 2023-06-14:
@jadesjardins ?

jadesjardins, 2023-06-14:
I have not looked at the Matlab lossless scripts in a long time... and I could definitely picture that the re-referencing would only happen at the end of a sequence of channel flagging procedures in a row. That being said, I would recommend implementing the feature of re-referencing after every channel rejection flagging procedure individually. I agree that it is likely to have little impact (so can be low priority), but it would be the correct thing to do.