lina-usc/pylossless #135: IC SD2 from MATLAB version

Andesha, 2023-06-08:
Working data, I've been seeing that the IC SD2 from the MATLAB version really helps to mark out those last few periods of artifacts.

We currently only do the one pass, but it does exist in the previous.

See [here](https://github.com/BUCANL/EEG-IP-L/blob/a56520698ec87bca2fd8897ac248454e03a35add/code/scripts/s05_concat_data.htb#L151-L154).

scott-huberty, 2023-06-08:
So just to clarify. In this case:

We would run the **_final_** ICA (in our case info max).

We run `flag_epochs_ic_sd` on the final ICA time series, where any flagged epochs are labeled `ic_sd_2`

This last assessment would not have any impact on the ICA topos obv because we aren't running another ICA afterwards.

The only impact of this assessment is to flag more epochs in the raw data, with `bad_pylossless_ic_sd2`

Is that right?

Andesha, 2023-06-13:
Yep, that's right.

It's just doing an additional flagging of bad time via exactly what you outlined

scott-huberty, 2023-06-13:
Okay cool, if you'd like this step and it was in the original pipeline Im +1 for adding this extra step.

@Andesha if you'd like this feature are you up for a PR? I won't have time to implement it myself in the near future.

I think we can just re-use the `flag_epoch_ic_sd` or whatever it's called and place it after the final ica. We can  create a parameter like `flag_label` with default `ic_sd1` but for this new second call we can change it to something else. This flag_label parameter should get passed in to `add_pylossless_annotation`

I think there are a few nested funcs that will need this new `flag_label` kwarg but seems do able to me.

Andesha, 2023-06-14:
I'll give it a try at some point - could you create it as an issue and assign it to me?

scott-huberty, 2023-06-14:
> I'll give it a try at some point - could you create it as an issue and assign it to me?

Done ;)

christian-oreilly, 2024-12-09:
I may be wrong, but I think this issue and #111 are related.