lina-usc/pylossless #157: Loading of flags is empty

Andesha, 2024-02-12:
Currently when you do a load ll derivative, it prints out a bunch of nicely formatted stuff about the flags. However, once you try and access these via `flags['ch']` , you'll find the structure empty

scott-huberty, 2024-02-12:
That's not good. Any chance you can share a file (or point me to its location on a CC cluster) so that I can take a look?

christian-oreilly, 2024-11-12:
Running the following code to try to reproduce the issue
```python

from pylossless.datasets import load_openneuro_bids
from pathlib import Path
import shutil
from mne import Annotations
import pylossless as ll

raw, config, bids_path = load_openneuro_bids()
# raw.crop(tmin=0, tmax=60)  # Too short for ICA to converge in some tests.
annots = Annotations(
    onset=[1, 15], duration=[1, 1], description=["test_annot", "test_annot"]
)
raw.set_annotations(annots)

config["find_breaks"] = {}
config["find_breaks"]["min_break_duration"] = 9
config["find_breaks"]["t_start_after_previous"] = 1
config["find_breaks"]["t_stop_before_next"] = 0
config["ica"]["ica_args"]["run1"]["max_iter"] = 5000

# Testing when passing the config object directly...
pipeline = ll.LosslessPipeline("test_config.yaml", config)

not_in_1020 = ["EXG1", "EXG2", "EXG3", "EXG4", "EXG5", "EXG6", "EXG7", "EXG8"]
pipeline.raw = raw.pick("eeg", exclude=not_in_1020).load_data()
pipeline.run_with_raw(pipeline.raw)

print("BEFORE:", pipeline.flags['ch'])
bids_path.root = bids_path.root / "derivatives" / "pylossless"
pipeline.save(bids_path, overwrite=True, format="EDF", event_id=None)
pipeline_loaded = ll.LosslessPipeline().load_ll_derivative(bids_path)
print("AFTER:", pipeline_loaded.flags['ch'])

shutil.rmtree(bids_path.root)

```

I get 
```
BEFORE: Flagged channels: |
  Noisy: None
  Bridged: None
  Uncorrelated: None
  Rank: ['Oz']
```

and 

```
AFTER: Flagged channels: |
  Noisy: None
  Bridged: None
  Uncorrelated: None
  Rank: ['Oz']
```

I would tend to conclude from that that this issue was somehow resolved by past PR. Unless I did not get the issue right?

@Andesha @scott-huberty Do you agree that this code demonstrate that this issue has already been solved?

scott-huberty, 2024-11-12:
Yeah.. checks out to me. @Andesha can you let us know if we are missing something since you opened this issue?

christian-oreilly, 2024-11-13:
If I hear counterindication, I will close with the merge of PR #178 which in the end do fix some stuff regarding the flag loading, but for the Epochs... also add tests that validate the load for ch and epoch flags.

PR #178 (2024-11-13): Adding test for flags after loading.
Closes #157