lina-usc/pylossless #162: How many versions of MNE should we officially support?

scott-huberty, 2024-07-22:
Originally posted in https://github.com/lina-usc/pylossless/pull/161#issue-2423857408 :

Since v1.7 (April 2024), MNE uses [edfio](https://github.com/the-siesta-group/edfio) for writing EDF files. So we no longer should depend on edflib-python, but should use edfio.

There are a couple of ways we can go about this.

1. Add a lower cap on the MNE version in our requirements.txt file, i.e. `mne>=1.7`. This assures that people don't have an older version of MNE installed that will try to use edflib-python.
2. Follow [SPEC0](https://scientific-python.org/specs/spec-0000/) and officially support all MNE versions that were released within the last 2 years (so that goes back to mne 1.4). Thus, we would need to add code that checks what MNE version the user has installed. If it is `<1.7`, check if `edflib-python` is installed. If it is `>1.7`, check if `edfio` is installed, etc.

Or I guess we could do a middle ground and continue to support mne v1.6, and thus add code to check if `edflib-python` or `edfio` needs to be installed, as previously described.

Personally I vote for 1, because I think the maintenance burden of 2 is currently too much for this project. Also, MNE only officially supports their latest stable release, so I don't want to support people using older versions that MNE doesn't even support themselves.

@Andesha @jadesjardins @christian-oreilly What do you think?

christian-oreilly, 2024-07-25:
@scott-huberty : Duplication of what I answered in #161 

> SPEC0 is a nice initiative. I did not know about it. In general, I favor the requirements being as flexible as possible to avoid creating dependency conflict in downstream projects but I am good with 1 too because this is a relatively young project so I think we need to keep the maintenance burden relatively low until there is more buy-in from the community.

christian-oreilly, 2024-11-17:
@scott-huberty Should we close this issue with a PR updating the requirements for `mne>=1.7`?

PR #188 (2024-11-19): Update requirements.txt with mne>=1.7 (#162)
Close #162
closed 2024-11-19 by christian-oreilly