lina-usc/pylossless #165: Release .2

scott-huberty, 2024-07-26:
There have been a number of bug fixes and enhancements since the initial release, so I think now is a good time to cut a new release. I've started a [project tracker](https://github.com/orgs/lina-usc/projects/3/views/1). I'm thinking September 1st is a generous timeline. I'm also going to be processing some data in August and so I will take the opportunity to look out for anything that should be improved. @christian-oreilly does this sound reasonable to you?

christian-oreilly, 2024-07-26:
Sure. Not all open issues are associated with a release. Maybe we should go through those that are not associated and attach them to a tentative release? It can be .3 or later if you don't want to include them in .2... and we can adjust these down the road... but having them labeled may help provide clarity about the objectives of the forthcoming releases.

scott-huberty, 2024-07-26:
Sure! I recently assigned some to .3, but feel free to swap tickets between .3 and .2 as you see fit.

christian-oreilly, 2024-11-15:
@scott-huberty `StLaurent`'s fine for the 0.2 release name?

scott-huberty, 2024-11-15:
sure :)

Question: If-when we release minor/patch releases do we just fall back to SemVer for the name of those releases (e.g 0.2.1)?

christian-oreilly, 2024-11-17:
> Question: If-when we ship minor/patch releases do we just fall back to SemVer for the name of those releases (e.g 0.2.1)?

Seems reasonable to me!

christian-oreilly, 2024-11-17:
PyLossless v0.2.0 released and deployed on PYPI! 🎆

closed 2024-11-17 by christian-oreilly

scott-huberty, 2024-11-17:
Woohoo!!!