lina-usc/pylossless #170: Saving the processed data from the pylossless

Deepa-Tilwani, 2024-10-04:
My code crash is happening while exporting it to bids format. I defined a bids path like this :

from mne_bids import BIDSPath
output_path = '/Users/deepatilwani/Documents/Projects/ABC/bids'
bids_path = BIDSPath(subject=df_paths["eeg_fname"][1].split(".")[0], task='annotation',
                 extension= '.edf', datatype='eeg', root=output_path)

It creates a path like this :
BIDSPath(
root: /Users/deepatilwani/Documents/Projects/ABC/bids
datatype: eeg
basename: sub-ABC1064_task-annotation)

When I do pipeline.save(bids_path)

It crashes
error message

It says :
{
	"name": "FileExistsError",
	"message": "\"/Users/deepatilwani/Documents/Projects/ABC/bids/sub-ABC1064/eeg/sub-ABC1064_task-annotation_events.tsv\" already exists. Please set overwrite to True.",
	"stack": "---------------------------------------------------------------------------
FileExistsError                           Traceback (most recent call last)
Input In [123], in <cell line: 1>()
----> 1 pipeline.save(bids_path)

File ~/Documents/Projects/venv/lib/python3.9/site-packages/pylossless/pipeline.py:1041, in LosslessPipeline.save(self, derivatives_path, overwrite)
   1031 def save(self, derivatives_path, overwrite=False):
   1032     \"\"\"Save the file at the end of the pipeline.
   1033 
   1034     Parameters
   (...)
   1039         whether to overwrite existing files with the same name.
   1040     \"\"\"
-> 1041     mne_bids.write_raw_bids(
   1042         self.raw,
   1043         derivatives_path,
   1044         overwrite=overwrite,
   1045         format=\"EDF\",
   1046         allow_preload=True,
   1047     )
   1048     # TODO: address derivatives support in MNE bids.
   1049     # use shutils ( or pathlib?) to rename file with ll suffix
   1050 
   1051     # Save ICAs
   1052     bpath = derivatives_path.copy()

File <decorator-gen-464>:12, in write_raw_bids(raw, bids_path, events, event_id, anonymize, format, symlink, empty_room, allow_preload, montage, acpc_aligned, overwrite, verbose)

File ~/Documents/Projects/venv/lib/python3.9/site-packages/mne_bids/write.py:1983, in write_raw_bids(raw, bids_path, events, event_id, anonymize, format, symlink, empty_room, allow_preload, montage, acpc_aligned, overwrite, verbose)
   1979 events_array, event_dur, event_desc_id_map = _read_events(
   1980     events, event_id, raw, bids_path=bids_path
   1981 )
   1982 if events_array.size != 0:
-> 1983     _events_tsv(
   1984         events=events_array,
   1985         durations=event_dur,
   1986         raw=raw,
   1987         fname=events_tsv_path.fpath,
   1988         trial_type=event_desc_id_map,
   1989         overwrite=overwrite,
   1990     )
   1991     _events_json(fname=events_json_path.fpath, overwrite=overwrite)
   1992 # Kepp events_array around for BrainVision writing below.

File ~/Documents/Projects/venv/lib/python3.9/site-packages/mne_bids/write.py:327, in _events_tsv(events, durations, raw, fname, trial_type, overwrite)
    324 else:
    325     del data[\"trial_type\"]
--> 327 _write_tsv(fname, data, overwrite)
[... 109 lines cut ...]
    151 # set patient info
    152 subj_info = raw.info.get(\"subject_info\")

File ~/Documents/Projects/venv/lib/python3.9/site-packages/edfio/edf_signal.py:124, in EdfSignal.__init__(self, data, sampling_frequency, label, transducer_type, physical_dimension, physical_range, digital_range, prefiltering)
    122 if not np.all(np.isfinite(data)):
    123     raise ValueError(\"Signal data must contain only finite values\")
--> 124 self._set_physical_range(physical_range, data)
    125 self._set_digital_range(digital_range)
    126 self._set_data(data)

File ~/Documents/Projects/venv/lib/python3.9/site-packages/edfio/edf_signal.py:427, in EdfSignal._set_physical_range(self, physical_range, data)
    425     data_max = data.max()
    426     if data_min < physical_range.min or data_max > physical_range.max:
--> 427         raise ValueError(
    428             f\"Signal range [{data_min}, {data_max}] out of physical range: [{physical_range.min}, {physical_range.max}]\"
    429         )
    430 self._physical_min = encode_float(
    431     _round_float_to_8_characters(physical_range.min, math.floor)
    432 )
    433 self._physical_max = encode_float(
    434     _round_float_to_8_characters(physical_range.max, math.ceil)
    435 )

ValueError: Signal range [0.0, 0.137605265721] out of physical range: [0.001804231242, 0.137605265721]"
}

scott-huberty, 2024-10-04:
1. For the first error, maybe we can improve our error message so that it is more clear to users what they need to do. I think that `'Please pass overwrite=True to the save method'`, might be better than `'Please set overwrite to True'`

scott-huberty, 2024-10-08:
This issue was fixed by updating to the current stable version of MNE-Python (version `1.8`). 

The reason this resolved the issue is because the bug that @Deepa-Tilwani was hitting originated from MNE-Python, not PyLossless, and that bug was fixed by MNE-Python in https://github.com/mne-tools/mne-python/pull/12676 , which was included in the `1.8` release.

closed 2024-10-08 by scott-huberty