lina-usc/pylossless #171: Utilize `desc` BIDS entity

scott-huberty, 2024-10-21:
BIDS/MNE-BIDS allows you to specify a `desc` entity for _derivatives_ data, to provide some description of the data. 

Utilizing this would allow us to add `desc-pylossless` or `desc-ica` to the filename, instead of doing:

`ica_bidspath = bpath.update(extension=".fif", suffix='_ica', check=False)`

Where we force a Non BIDS-compliant suffix.

However this technically will be API breaking so might require some discussion and extra communication to users.

christian-oreilly, 2024-10-28:
We could probably support both naming conventions on "read" to avoid breaking people's stuff while emitting a deprecation warning for people still reading files with the extension. I am not sure it has to break the API...

scott-huberty, 2024-10-28:
> We could probably support both naming conventions on "read" to avoid breaking people's stuff while emitting a deprecation warning for people still reading files with the extension. I am not sure it has to break the API...

I like your style!

scott-huberty, 2024-11-11:
Note to self: There is also a `proc` (short for processed) entity that might be more suitable. i.e. `sub-01_task-rest_proc-pylossless_eeg.edf`