lina-usc/pylossless #174: Config error dictionary update sequence element #0 has length 1; 2 is required

Deepa-Tilwani, 2024-11-11:
Bub on loading a config file 

config = ll.config.Config("abc_ll_config.yaml")
config.load_default()
#config.print()
config.save()

Error : 

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[6], [line 1](vscode-notebook-cell:?execution_count=6&line=1)
----> [1](vscode-notebook-cell:?execution_count=6&line=1) config = ll.config.Config("abc_ll_config.yaml")
      [2](vscode-notebook-cell:?execution_count=6&line=2) config.load_default()
      [3](vscode-notebook-cell:?execution_count=6&line=3) #config.print()

ValueError: dictionary update sequence element #0 has length 1; 2 is required

scott-huberty, 2024-11-11:
Thanks @Deepa-Tilwani - just a hint on a little formatting trick: if you wrap code and stack traces in backticks (i.e. "```" ) then GitHub will render them as code-blocks. I just edited your ticket to do this.

I can replicate this. @christian-oreilly it's probably related to the #173 . 
```python

import pylossless as ll
config = ll.config.Config("abc_ll_config.yaml")
```

Unfortunately the stack trace isn't very informative:

```console
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
    [... skipping hidden 1 frame]

Cell In[3], line 1
----> 1 config = ll.config.Config("abc_ll_config.yaml")
      2 config.load_default()

ValueError: dictionary update sequence element #0 has length 1; 2 is required

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
File <ipython-input-1-68856df000eb>:1
----> 1 config = ll.config.Config("abc_ll_config.yaml")

ValueError: dictionary update sequence element #0 has length 1; 2 is required
> <ipython-input-1-68856df000eb>(1)<module>()
----> 1 config = ll.config.Config("abc_ll_config.yaml")
```

 I'm kind of shocked that this did not come up in our tests. But it's because we never actually pass a custom filename to the `ll.config.Config` constructor in our tests.

So in at least one of our tests or documentation tutorials, we should do `ll.config.Config("my_special_config.yaml")`

scott-huberty, 2024-11-11:
Wait a second... False alarm 🚨 ...

@Deepa-Tilwani We have never supported passing a filename to the config constructor. Instead this is what you need to do.

```
config = ll.config.Config()
config.load_default()
config.save("ll_abc_config.yaml")
```

Can you confirm that this fixes it, and I'll close the issue.

Deepa-Tilwani, 2024-11-11:
This works @scott-huberty. Thanks.

closed 2024-11-11 by scott-huberty

scott-huberty, 2024-11-12:
@Deepa-Tilwani I think that we must have mixed up the API for the lossless pipeline constructor, where you do indeed pass the config filename:

`pipeline = ll.LosslessPipeline('my_project_ll_config.yaml')`

I think if you were just trying to read the `config` file from disk, this might work:

`config = ll.config.Config()`
`config.read("ll_abc_config.yml")`

christian-oreilly, 2024-11-12:
For a moment, I was puzzled about why we did not allow the construction of the Config class to take as an input a config path. It seems like an obvious thing. Looking into it, this is because we modeled the config class by derivating it from the dict class. So, that class is mostly just a dictionary class to which we added functions to load and save YAML files... I guess we could still implement a constructor for it that breaks from the behavior of the dict constructor but that could have some side effects...

scott-huberty, 2024-11-12:
Yeah, Happy to consider a PR if @Deepa-Tilwani would find this useful.