lina-usc/pylossless #183: pylossless_qc fails to launch the QC dashboard

christian-oreilly, 2024-11-14:


scott-huberty, 2024-11-14:
Do you mean that, the dashboard doesn't open in the browser _automatically?_

christian-oreilly, 2024-11-17:
Sorry for the non-descriptive issue. I created these as I was on a call with a student who was trying to run PyLossless and it was not running. 

> Do you mean that, the dashboard doesn't open in the browser automatically?

Nope. I created #182 for that. I meant that `pip install --editable .[dash]` while being in the cloned repo installed pylossless fine, except that the command `pylossless_qc` was not recognized by the system. We did not have much time to investigate this specifically on the call, so I would need to check if I can manage to reproduce this issue to pinpoint the cause.

scott-huberty, 2024-11-17:
ah OK.

if it's of any help, I just install Pylossless in a fresh virtual environment and `pylossless_qc` worked fine for me:

Sounds like it could have been a Python PATH problem?

-----------------------------------------------------------------------------------------------------------------------

```bash

conda create -n ll_test python=3.10
conda activate ll_test
```

```bash
# confirm that python path is to the virtual env
which python
```

```bash
python -m pip install --editable ".[dash]"
```

```
pylossless_qc
```
```console
Dash is running on http://127.0.0.1:8050/

 * Serving Flask app 'pylossless.dash.app'
 * Debug mode: on
```

christian-oreilly, 2024-11-21:
Playing around with this installation, we got this error message:

[image]

So... as far as I can tell, this seems not to be related to PyLossless at all but some issue with system installation. We added the corresponding binary path to the PATH system variable in .bashrc. I'll close this issue as it does not seems specific to the PyLossless.

closed 2024-11-21 by christian-oreilly