lina-usc/pylossless #192: HPC-friendly PyLossless

christian-oreilly, 2024-12-03:
The aim of this issue is to merge back in the main PyLossless project the modifications that were made in this fork https://github.com/Andesha/pylossless to make PyLossless more HPC-friendly. My understanding from the diff is that most of these modifications have to do with making the package more lightweight, but most importantly, reducing dependencies (e.g., removing dependencies related to the dashboard, which is not run through bash scripts used in an HPC context). I think the functionalities required by this fork could be implemented as part of the main PyLossless project with a install of the type `pip install pylossless[light]`

christian-oreilly, 2024-12-03:
@Andesha, please feel free to comment if you think I am not capturing your specific requirements or if you think this issue is a bad idea overall. I discussed this with @jadesjardins earlier today.

christian-oreilly, 2024-12-09:
Comparing the fork https://github.com/lina-usc/pylossless/compare/main...Andesha:pylossless:main

71 files are modified. 

Among those, the following files were removed:
- 4 files from`.github/workflows`.
- .pre-commit-config.yaml, .readthedocs.yaml, and codecov.yml. 
- 5 files in the `examples` folder. 
- 2 files in the `notebooks` folder.
- 28 pylossless/assets/... files
- 13 pylossless/dash/... files
- 3 pylossless/dataset/... files
- 3 pylossless/tests/... files
- removal of requirements_qc.txt and requirements_texting.txt

I don't think that the presence of these files would actually cause an issue for the execution on HPC. It may make the package slightly larger, but this effect should be fairly negligible. If some of these files were removed because their functionalities require dependencies not wanted for an HPC application, I think the issue needs to be solved with the installer script, not be removed files from the package. So, I propose not merging these changes.

One additional change not to be merged:
- Change to .gitignore to add a `test.py` to the files to be ignored.

Here are the files which diff need to be further considered to merge relevant changes:
- README.md
- __init__.py
- pylossless/assets/ll_default_config.yaml
- pylossless/flagging.py
- pylossless/pipeline.py
- requirements.txt
- setup.py

christian-oreilly, 2024-12-09:
The requirements from @Andesha has `numpy<2`. I'd rather not include that. Less-than version requirements are nasty. To be avoided if possible. Do we know if this is still needed and if so, what is the context?

christian-oreilly, 2024-12-09:
@scott-huberty Part of the changes were substituting  
```python
            f"  Noisy: {self.get('ch_sd', None)}\n"
            f"  Bridged: {self.get('bridge', None)}\n"
            f"  Uncorrelated: {self.get('low_r', None)}\n"
```

by 

```python
            f"  Noisy: {self.get('noisy', None)}\n"
            f"  Bridged: {self.get('bridged', None)}\n"
            f"  Uncorrelated: {self.get('uncorrelated', None)}\n"
```

On the one hand, the latter is definitely more readable. On the other hand, it is a breaking change. What is your opinion on adopting that change?

christian-oreilly, 2024-12-09:
For pylossless/assets/ll_default_config.yaml, Taylor's fork scrubs slurm, general_info, and t_info from the config file. I cannot see these fields used anywhere in the current code base, so I tend to align with his choice to remove these.

scott-huberty, 2024-12-09:
> For pylossless/assets/ll_default_config.yaml, Taylor's fork scrubs slurm, general_info, and t_info from the config file. I cannot see these fields used anywhere in the current code base, so I tend to align with his choice to remove these.

Agreed.

scott-huberty, 2024-12-10:
> @scott-huberty Part of the changes were substituting
>
> ```python
>             f"  Noisy: {self.get('ch_sd', None)}\n"
>             f"  Bridged: {self.get('bridge', None)}\n"
>             f" [...]

I'm sorry I'm not following. Here is the pattern on main:

https://github.com/lina-usc/pylossless/blob/c647153bd39315b94f39a122a2e8629a3c1f08d5/pylossless/pipeline.py#L507-L512

In the code you are displaying, it seems like f strings are being used instead of a dictionary.

BTW If by breaking change you are referring to the flag names, That was merged in https://github.com/lina-usc/pylossless/pull/152

christian-oreilly, 2024-12-10:
Well... it seems like pipeline.py and flagging.oy as not in sync somehow... or I am missing something... https://github.com/lina-usc/pylossless/blob/c647153bd39315b94f39a122a2e8629a3c1f08d5/pylossless/flagging.py#L21-L26

christian-oreilly, 2024-12-10:
OK, it seems that what it is only the __repr__ that was not properly updated by #152 originally. Adjusting the flagging.py consistently with 
```python
 # Flagged Channels 
 flagged_channels_data = { 
     "Noisy": ch_flags.get("noisy", None), 
     "Bridged": ch_flags.get("bridged", None), 
     "Uncorrelated": ch_flags.get("uncorrelated", None), 
 } 
```

christian-oreilly, 2024-12-10:
For the record, the approach with `pip install pylossless[light]` would not work because the `extra` in `pip install package[extra]` can only be used to add **extra** dependencies, not alternative dependencies... so it cannot be used to install less dependencies. We would need to have "light" by default and use something like `[full]` to have all dependencies... which I think is not desirable. Anyway, looking at the diff, I don't think it is need anyway.

scott-huberty, 2024-12-10:
Right... It depends on what dependencies are the issue.

@Andesha Can you please let us know.

christian-oreilly, 2024-12-10:
Yeah. I integrated whatever changes seem to make sense of it and I was thinking of asking @jadesjardins to check if the PR works for him because it is my understanding that he needs that for running Q1K through the pipeline. I guess @Andesha is probably using his fork for wrapping pylossless for EEGNet, so the requirement may be a bit different. If that is the case, maybe you could also test it, @Andesha, and clarify what needs to be edited to make it work.

jadesjardins, 2024-12-10:
I am still getting the same error when trying to run the hpc-friendly-pylossless branch on Narval... this is the error message:

$ cat slurm-38223405.out
Traceback (most recent call last):
  File "/lustre06/project/6030602/q1k/experimental/HSJ/derivatives/pylossless/code/q1k_submit_pylossless/main.py", line 1, in <module>
    import pylossless as ll
  File "/project/def-emayada/q1k/experimental/q1k_env/lib/python3.11/site-packages/pylossless/__init__.py", line 1, in <module>
    from . import pipeline, bids, config, datasets, flagging, utils
  File "/project/def-emayada/q1k/experimental/q1k_env/lib/python3.11/site-packages/pylossless/pipeline.py", line 33, in <module>
    from .flagging import FlaggedChs, FlaggedEpochs, FlaggedICs
  File "/project/def-emayada/q1k/experimental/q1k_env/lib/python3.11/site-packages/pylossless/flagging.py", line 16, in <module>
    import mne_icalabel
  File "/project/def-emayada/q1k/experimental/q1k_env/lib/python3.11/site-packages/mne_icalabel/__init__.py", line 5, in <module>
    from . import features  # noqa: F401
    ^^^^^^^^^^^^^^^^^^^^^^
  File "/project/def-emayada/q1k/experimental/q1k_env/lib/python3.11/site-packages/mne_icalabel/features/__init__.py", line 3, in <module>
    from .topomap import get_topomaps  # noqa: F401
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/project/def-emayada/q1k/experimental/q1k_env/lib/python3.11/site-packages/mne_icalabel/features/topomap.py", line 6, in <module>
    from mne.io import Info
ImportError: cannot import name 'Info' from 'mne.io' (/project/def-emayada/q1k/experimental/q1k_env/lib/python3.11/site-packages/mne/io/__init__.py)

christian-oreilly, 2024-12-10:
OK. Thanks, this is useful. I did not know this was the issue. Let me check the issue and push a commit.

christian-oreilly, 2024-12-10:
I think this bug was present in mne_icalable v0.4 and fixed in v0.5. Currently, the requirement file set "mne_icalabel>=0.6.0", so it should be fine. How do you install pylossless?

christian-oreilly, 2024-12-10:
In the section to install for HPC, the readme on the master install specifically v0.4, which would cause the issue. On this branch, the corresponding text is 
```python
# Clone down mne-iclabel and switch to the right version and install it locally
cd ~/eeg-env
git clone https://github.com/mne-tools/mne-icalabel.git
cd mne-icalabel
git checkout maint/0.5
pip install .
```
So it should work... but do we have any reason to believe it is actually necessary to install this from source? Could you try to see if `pip install --upgrade mne-icalabel` fix your issue?

jadesjardins, 2024-12-10:
in my test environment I definitely used mne-icalabel main/0.4 .... I will try 0.5 and let you know.

jadesjardins, 2024-12-10:
actually, I will try the pip install --upgrade first...

christian-oreilly, 2024-12-10:
Unless there were some HPC-specific issues with using the latest version of that package (0.7), the pip install --upgrade should work fine! And is simpler... I expect that the installation from the clone was used at some point where a recent PR was necessary, but was not yet merged in the latest release. Such PR has long been included since they made multiple releases since then.

jadesjardins, 2024-12-10:
it is working after the upgrade!

scott-huberty, 2024-12-10:
Thanks @christian-oreilly for the detective work 🕵️

christian-oreilly, 2024-12-10:
Alright! I changed the README to use the current version of mne-icalabel instead of an outdated clone. I'll give a few days for @Andesha to comment if he cares to comment about the usability of this version for his application (EEGNet I suppose) and will proceed with these changes if he does not comment (I think he is probably working on other stuff these days).

PR #198 (2024-12-12): Merging code from Taylor's fork
Closes #192
closed 2024-12-12 by christian-oreilly