lina-usc/pylossless #208: Creating Annotation Manual For Pylossless

iofeii, 2024-12-12:
This ticked is for creating the manual for the quality control annotation process in Pylossless.

scott-huberty, 2024-12-12:
Hello @iofeii - Welcome!

Can you provide more information on what this ticket is about? 

- Are you having an _issue_ with the PyLossless Quality Control Dashboard that you would like to fix?
- Is there a new _feature_ that you would like to request, that involves manually creating annotations?

iofeii, 2024-12-12:
Hello! This ticket was created to begin creating instructional documentation for manual annotation in the Pylossless QC Dashboard as part of my undergraduate research tasking under Dr. O’Reilly.

scott-huberty, 2024-12-12:
Ok :)

christian-oreilly, 2024-12-12:
I updated the issue description to clarify a bit more the deliverable.