lina-usc/pylossless #218: The signal colors on the dashboard are indistinguishable

iofeii, 2025-01-28:
The current colors of the web dashboard make it difficult to quickly differentiate between different signal types. 

[image]

It may be beneficial to look at other color palettes that would be more distinguishable.

scott-huberty, 2025-01-28:
+1 :) 

Let's mimic one of the available seabird color_pallettes? Or one provided by Plotly: https://plotly.com/python/discrete-color/

This is a great blog about how the Matplotlib devs landed on their default cmaps: https://bids.github.io/colormap/

iofeii, 2025-02-11:
Below are the proposed new signal colors for the plots as well as an example of their implementation with a data set. The used data set does not include all the signals, but it does include signals whose corresponding colors are the closest/least distinguishable of the group. This new scheme also include a change in the BAD signals area (both automatically and manually designated) as it is a lighter shade of red than before. This ensures that all signal colors are distinguishable from the background. 

The background of the topo map area will need to be lighter (perhaps just a lighter gray) for better visibility of the outlining colors and labels. 

**In pylossless/dash/__init__.py**

IC_COLORS = [
    "#2c2c2c",  # Brain, Dark Gray
    "#0173b2",  # Muscle, Blue
    "#8d68b0",  # EOG, Lavender
    "AA336A",   # ECG, Dark Pink
    "#ffcc00",  # Channel Noise, Dark Yellow
    "#7d7f7c",  # Line Noise, Medium Gray
    "#009696",  # Other, Teal
]

[image]

[image]

scott-huberty, 2025-02-11:
Thanks @iofeii !

Do you have a source for your color choices? E.g. a pre-existing discrete colormap used by Matplotlib / Seaborn / Plotly, or some literature on why it is a good choice.

scott-huberty, 2025-02-11:
One problem is that component signals that are deemed "to be excluded"  should be turned light gray in color. This may not be implemented yet but should/will be. So we don't want to "line noise" to be Medium gray because that will be confusing

I suggest us not to go rogue here (i.e. avoid the mistake I made the first time I set these colors). If we are concerned about colors being distinguishable then let's use a well established color palette:

For example `sns.color_palette("colorblind")`

christian-oreilly, 2025-02-20:
@scott-huberty I would just point out that perceptual uniformity is not a relevant property for categorical coloring. It makes sense for a "scale" to be uniformly mapped to a color scheme so that however different two colors seem to be from one another matches however large is the difference between two numbers that they encode. But when color-coding categories (apples, tomatoes, and bananas), there is no measure of "distance" between categories so the perceived difference in colors does not need to match a distance and the perceptual uniformity criterion does not make sense anymore. In that context, you just need the colors associated with the categories to be maximally distinguishable. I guess picking colors equally spaced on a "perceptually uniform" color map could maximize the differentiability, provided that the range of perceptual changes is maximal (e.g., I think viridis has a larger range than crest as the range of color variation in crest seems to be more or less a subset of the one in viridis).

scott-huberty, 2025-02-20:
Maybe I misused the term "perceptually uniform" there, but I think I disagree that perceptual distance does not matter for categorical palettes -  the doc I linked to does include the "Qualitative" palettes (`tab10`, `color_blind`, `Dark2` etc), and for me the general point is that considerable thought went into designing those palettes so that they are still distinguishable even if converted to grayscale (or perhaps more importantly, they are viewed by someone with severe color vision deficiency) 

That is my 2c, but If I am the sole blocker on this moving forward, @iofeii  please feel free to open up a PR with your proposed color palette!

christian-oreilly, 2025-02-20:
Sure. Qualitative palettes work well for this kind of application. I agree with the general idea of reusing what people spent a lot of time thinking about instead of going from scratch. I'd be mindful about the reuse though so that it fits with our use case (e.g., color-blindness is important; mapping to greyscale is not since dashboards are not meant for printing) and that it does not compromise on the ease of distinguishing colors to address a problem we don't have in our specific use case (e.g., greyscale mapping).

iofeii, 2025-04-11:
This is the latest update for the color palette choices as the working document for the manual has been shared and includes these decisions. I have taken all we've discussed into consideration and decided to base it upon the Seaborn Colorblind palette. I did consider this palette before; however, I initially was against it because the color palette clashed with the background color of the BAD_LL regions. To be specific, oranges and yellows which were key contributors to the palette were hidden in the harsh red background (Fig. 1). Dr. O'Reilly suggested choosing another color for those regions. Keeping in mind that the red was originally chosen to portray a negative connotation (for "noisy"/"BAD), I decided to change to color to a lighter red ("Pastel Red", #FAA0A0). With this lighter red, the Seaborn Colorblind palettes do not clash as aggressively. However, it is important to note that this red is not originally in the Seaborn Colorblind palette as it was handpicked from other sources which would go against the goal of sticking with supported documentation. There is no "red" in the Seaborn Colorblind palette. The closest there is would be "Bright Pink" (#fbafe4) which may not suit our needs as it almost masks the purple signals. Let me know if would have further insight on that (Fig. 2). As for the signal colors, all the colors come from the Seaborn colorblind palette except for the "brain". The brain, I kept as black for easy recognition. Let me know your opinions on that. This black decision is seen on Figure 2 and Figure 3. 

Below are the 10 signals with their new associated colors in the IC_COLORS dictionary within __init__.py:

    "#000000", # BRAIN, Black
    "#0173b2", # Muscle, Dark Blue
    "#cc78bc", # EOG, Purple
    "#de8f05", # ECG, Orange
    "#56b4e9", # LINE NOISE, Cyan
    "#ca9161", # CHANNEL NOISE, Brown
    "#029e73", # OTHER, Green
    "#d55e00",  # Unclassified, Reddish-Brown
    "#fbafe4",  # Unclassified, Bright Pink
    "#ece1334",  # Unclassified, Yellow

By default, there are 10 colors in the Seaborn Colorblind Palette. Only 9 of the original 10 are used as of now. The remaining color is "grey" and has been switched out for black as previously stated. In addition, we already used a grey to designate BAD lines. 

Please let me know if you have further suggestions on any of the color map decisions that have been made and are currently portrayed on the working manual. I am more than happy to share more example photos of changes and other implementations for this issue. 

Examples:
**Figure 1: Default Seaborn Colorblind Palette on Unchanged BAD_LL Background**

[image]

**Figure 2: Seaborn Colorblind Palette on Bright Pink BAD_LL Background**

[image]

**Figure 3: Seaborn Colorblind Palette on Pastel Red BAD_LL Background**

[image]

scott-huberty, 2025-04-16:
> There is no "red" in the Seaborn Colorblind palette.

True but there are reds in other Seaborn categorical palettes that you can pull from if it is helpful:

https://seaborn.pydata.org/tutorial/color_palettes.html