lina-usc/pylossless #220: run_with_raw(raw) not saving derivatives by default

Deepa-Tilwani, 2025-02-07:
[image]

Expected Behavior:

The run_with_raw(raw) method should default to saving derivatives, similar to the run method, which saves files when save=True

christian-oreilly, 2025-02-07:
@Deepa-Tilwani When you want to save, at this stage, you want to save the files in their "lossless" state (annotated, not cleaned yet) or do you want to save the cleaned lossy version? 

Let's clarify this before I comment on @scott-huberty's message. 

BIDSPaths are a pain to work with. I wish we could find a more straightforward way to manage these files. For example, for the very basic use-case "I have a raw file, I want to clean it and save the cleaned raw files as cleaned.edf", the user should not have to do 40 different steps...

Deepa-Tilwani, 2025-02-07:
@christian-oreilly I'm working on our valence project. The `Run_dataset` function will save it to the lossless version, but if some files are missing in the lossless format, it should then check for the BIDS format. However, it should skip looking for the BIDS format if I already have a raw EDF file loaded, it should read_with_raw. And when we do read_with_raw it is not saving lossless version. 

For now, I am using the bids format, which works fine. But if someone wants to save the lossless version from raw with the pylossless, then it should be saved in the derivatives path.

scott-huberty, 2025-02-07:
> BIDSPaths are a pain to work with. I wish we could find a more straightforward way to manage these files. For example, for the very basic use-case "I have a raw file, I want to clean it and save the cleaned raw files as cleaned.edf", the user should not have to do 40 different steps...

I completely agree.