lina-usc/pylossless #225: Add a tutorial on how to launch and use the QC dashboard

scott-huberty, 2025-03-10:
@christian-oreilly it has been requested by a reviewer to provide more thorough documentation on how to launch and use the QC dashboard. IIRC you have a student working on this?

christian-oreilly, 2025-03-11:
@iofeii Where are you on this? I think you added whatever you could at this point? Could you prepare a PR so that we can integrate what you prepared so far and see if anything else need to be added for the this issue?

iofeii, 2025-04-01:
Here is the link to view the manual as is. It is currently in its working document state. A refined copy will be available by early May. 
https://docs.google.com/document/d/1xO5PtckM_SP0QywHy0JRub32DMlVE9IL2CQGfhmLonc/edit?usp=sharing

iofeii, 2025-04-01:
Hi, @scott-huberty, as of now, I couldn't find any other way to delete BAD regions/shapes besides using the built in "delete active shape" option on the top right of the plots. Just checking, is there a call-back/action that serves as another way to remove these regions?

christian-oreilly, 2025-04-02:
> Once the content is ready, I can give you tips on how to transfer this to a .py file that can be generated to sphinx documentation by sphinx-gallery (If that is something that you are interested in learning, and have the time for).

Funny you should say that, we were just discussing yesterday @iofeii working on that over the summer... :)